bigtreetech/BIQU-BX #4: Unable to Recompile Firmware

dastolly, 2021-03-06:
Cannot compile firmware with a successful flash, even without making any changes to the downloaded files.  Using VS.  I'm trying to get the USB on the TFT to be the connection to Octoprint, as well as enable safe home, but I cannot make these changes and have the printer successfully load the new firmware.  It runs the process up to 100%, but never boots, just stays at a black screen.

inismor, 2021-03-09:
> Cannot compile firmware with a successful flash, even without making any changes to the downloaded files. Using VS. I'm trying to get the USB on the TFT to be the connection to Octoprint, as well as [...]

Are you compiling BTT_SKR_SE_BX_BOOT? (Not BTT_SKR_SE_BX). Worked for me.

dastolly, 2021-03-11:
> Are you compiling BTT_SKR_SE_BX_BOOT? (Not BTT_SKR_SE_BX). Worked for me.

I'm compiling in Visual Studio and I can open 3 folders from the downloaded code.  One folder says "BIQU-BX", one says "BIQU-BX-1", and one says "BIQU-BX-master".  All three folders have similar innards, and all reside inside a "BIQU-BX-master" folder.  No matter which one I select, when I build, the marline auto build shows BTT_SKR_SE_BX.  How can I get to the boot one?

inismor, 2021-03-11:
What you describe isn’t what I saw, and I just downloaded the sources a couple of days ago. PlatformIO only shows BTT_SKR_SE_BX and BTT_SKR_SE_BX_BOOT. Build the _BOOT version and copy the resulting firmware.bin file to an SD card. First time you flash the printer, be sure to add the “config.ini" file and “BTT70" folder from the top-level "Copy to microSD to update” folder. BTT has made some changes to the UI since they shipped.

—r

dastolly, 2021-03-11:
> What you describe isn’t what I saw, and I just downloaded the sources a couple of days ago. PlatformIO only shows BTT_SKR_SE_BX and BTT_SKR_SE_BX_BOOT. Build the _BOOT version and copy the resulting [...]

I'll give it a shot again, thanks!  I'm just not seeing where to choose between the two.  I'll keep digging.

inismor, 2021-03-11:
You build with PlatformIO, right? Not the compile button.

dastolly, 2021-03-11:
In Visual?  I've been using auto build marlin... I'm in it now, trying to find it.  Thanks for sticking with me.

inismor, 2021-03-11:
Yeah, you have to use PlatformIO. It ensures that the proper libraries are present and linked into the code. The PlatformIO plug-in for VSCode is free.

—r