bigtreetech/BIQU-BX #5: SD card files

jtatum, 2021-03-10:
Is there a download somewhere that has all the SD card files? My SD card arrived blank/corrupted.

Will-wastelander, 2021-03-10:
I have been requesting this for a few weeks now.

looxonline, 2021-03-20:
There are links to it on their facebook group.

predy3dprint, 2021-03-20:
But not everybody has a Facebook Account :-(

thisiskeithb, 2021-03-20:
They’ve been uploaded: https://github.com/bigtreetech/BIQU-BX/tree/master/TF%20card%20data

thisiskeithb, 2021-03-22:
Here's a PR to add the original parts cooling duct STL: https://github.com/bigtreetech/BIQU-BX/pull/6

TRIBVNE, 2021-04-10:
> Is there a download somewhere that has all the SD card files? My SD card arrived blank/corrupted.

When you say "blank/corrupted" are you getting a read error?  I am getting a read error no matter what I try, and the only adapter I have to check it on my computer is the poor quality one that came with the BX.  It won't be recognized by my computer even if I try it with a card I have full of photos.

thisiskeithb, 2021-06-08:
https://github.com/bigtreetech/BIQU-BX/pull/13 adds the missing STL for the drawer.