bigtreetech/BIQU-BX #7: Using cura profile

grimlokason, 2021-03-22:
Hello, 

There is a cura profile to load, but no instruction on which printer we must use as a base.
Is it possible to add that instruction  ?

NB : with loading on a BIQU B1 in cura 4.8.0 we got an error : 
```Failed to import profile from C:/Users/grim_lokason/Downloads/BIQU-BX configuration file(1).curaprofile: Quality type 'normal' is not compatible with the current active machine definition 'biqu_base'.```

Best regards.

thisiskeithb, 2021-03-22:
Using Cura 4.8?

Will-wastelander, 2021-03-22:
I have tried with Cura 4.6, 4.7, and 4.8 and get an error when attempting
to the supplied profile.

looxonline, 2021-03-23:
> after more digging, the biqu_base.def.json is installed but not detected in the printer list in cura :/

The base definition is not a printer. It is a collection of settings that form a base for all BIQU machines to build off. A Cura profile is in development but since there has been no Cura release since Q3 2020 there has been no opportunity to have it integrated into upstream cura. If you would like to trial it and provide feedback you are more than welcome to do so by following the guide here: https://www.facebook.com/groups/1012996125813700/permalink/1156048964841748

hoyohayo, 2021-07-21:
Is there any way of being able to take part in the trial without having to use facebook?

I actually do not own an account, but would still like to participate and give feedback on that.

thisiskeithb, 2021-07-21:
> Is there any way of being able to take part in the trial without having to use facebook?

https://github.com/Ultimaker/Cura/pull/10086 was merged, so there will be a profile for the BX in Cura 4.11.