bigtreetech/BIQU-BX #8: Marlin configuration ?

grimlokason, 2021-03-22:
Hello,

There is a marlin folder on this projet, but i'm not able to find the config files for the biqu BX.

Is it possible to add them ?

There is the necessary for the B1 : https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/BIQU

Best regards.

thisiskeithb, 2021-03-22:
This printer is not supported upstream.

You’ll need to use the firmware from this repo as-is until it’s merged upstream and I’ll have a BX config ready similar to what I did for the B1.

grimlokason, 2021-03-23:
In fact, i was hopping too take a look at some conf information since i've a biqu H2 (the same extruder as in the BX if i'm right ),  on a ender 3v2 and didn't have great result.

If we take a look here : https://damsteen.nl/blog/2021/01/19/mounting-the-biqu-h2-to-creality-cr6 on "Other advanced settings"
He say to modify RETRACT_LENGTH / PAUSE_PARK_RETRACT_LENGTH / FILAMENT_CHANGE_UNLOAD_FEEDRATE / FILAMENT_CHANGE_UNLOAD_LENGTH / FILAMENT_CHANGE_FAST_LOAD_FEEDRATE  but i don't know how he choose those information :/

thisiskeithb, 2021-03-23:
> He say to modify RETRACT_LENGTH / PAUSE_PARK_RETRACT_LENGTH / FILAMENT_CHANGE_UNLOAD_FEEDRATE / FILAMENT_CHANGE_UNLOAD_LENGTH / FILAMENT_CHANGE_FAST_LOAD_FEEDRATE but i don't know how he choose those information :/

You can find that information in the BX config located in this repo: https://github.com/bigtreetech/BIQU-BX/blob/ce7c484c01f9e4bb9f87592a8fb6a1d4d184f583/Firmware/Marlin-2.0.6-BX/Marlin/Configuration_adv.h#L2045-L2098

grimlokason, 2021-03-23:
Thank you !
I don't know why i haven't been able to found it in the project.

closed 2021-03-23 by grimlokason