bigtreetech/BIQU-BX #10: Model gcode files

Protocol-X, 2021-04-11:
All of the models appear to be in a hex type format.   Please correct the format.   You have kit provided us start and end code to ret up these printers and there is no way to extract et from these files.  Trying to convert them to ascii just results in gibberish.

thisiskeithb, 2021-04-11:
The `.gcode` files have image preview data embedded so you have to scroll below that.

Example: For the current [3DBenchy.gcode](https://github.com/bigtreetech/BIQU-BX/blob/5c7d32e474446e1a88884869bf2cec28986cf9e2/TF%20card%20data/Models/3DBenchy.gcode), the Simplify3D gcode starts around 593:

[image]

Protocol-X, 2021-04-11:
Thank you for the information Keith

Protocol-X, 2021-07-26:
User error.

closed 2021-07-26 by Protocol-X