bigtreetech/BIQU-BX #14: Marlin-2.0.6-BX compiled firmware.bin results in black screen for TFT

R60D, 2021-08-11:
I was trying to flash my BIQU-BX with the latest compiled version using VSCODE AUTO BUILD MARLIN
[image]

Video of bricked BIQU-BX
https://user-images.githubusercontent.com/29761720/129068880-8d21c268-b605-46c7-90f6-5004df94bf8b.mP4

I was able to use the printer by flashing the pre-compiled firmware.bin that comes with the TFT icons, but I still cannot compile my own firmware.bin

R60D, 2021-08-12:
So I did some digging and found a repo that works for the BIQU-BX.

https://github.com/looxonline/Marlin/tree/BIQUBX_ALMOST_STOCK

I compiled the BTT_SKR_SE_BX_BOOT Enviroment. After flashing it to my printer, it worked like a charm. I had to remove the physical Z-endstop though as it is explained in the repo.
[image]

Gwebber-Net, 2022-01-07:
I have the same issue, can a developer answer ?

Mrjeanswe, 2022-05-10:
same problem , any solutions?

Gwebber-Net, 2022-05-10:
There is this firmware.
Go to the biqu bx Facebook page.     And look for the get started section.

You will find a GitHub page , with firmware for the stock machine

Outlook voor Android<https://aka.ms/AAb9ysg> downloaden

Rick18101965, 2022-05-10:
Not as yet thar i have heard...