bigtreetech/BIQU-BX #18: Too long extrusion prevented, when printing more then 1 model

Gwebber-Net, 2022-01-02:
When i am printing more then 1 model, always after a couple of layers, the error message pops up,
"Too long extrusion prevented".

And the most funny thing,
After that error, for some weird reason, my printer starts to print, an entirely different model, ontop of what is already printed.
So i was printing a Steppermount, and a benchy was printing halfway through.

radek8, 2022-01-04:
try to increase the value
#define EXTRUDE_MAXLENGTH

Gwebber-Net, 2022-01-05:
Yes, I figured that would help,. But I wasnt interested inin going into the firmware for a brand new printer.

I try to not tinker with it.

Outlook voor Android<https://aka.ms/AAb9ysg> downloaden