bigtreetech/BIQU-BX #22: Random restarts

AndyUK300ZX, 2022-02-27:
Is anyone experiencing random restarts after printer has been printing for over an hour?

Having finally got my BX printing well, any model that requires long print times fail. I've printed the usual files that came on the SD Card but once I started to print other files everything started going wrong.

None of the gcode files I create in Ultimaker Cura will complete a print. After an hour or so the print powers off then reboots. No print resume option even though print resume is on.

Is this a faulty machine or something in the code? I've used several different SD Card or USB stick all connected to the USB port on the TFT

petrbraha, 2022-02-27:
I have same issue, random restarts when printing from USB or SD card connected to TFT. When I print from microSD card connected to mainboard, it prints ok.

AndyUK300ZX, 2022-02-27:
> I have same issue, random restarts when printing from USB or SD card connected to TFT. When I print from microSD card connected to mainboard, it prints ok. Same G-Code at USB, SD and microSD.

That's helpful to know, thanks. Guess there must be an issue with the data pass through from the TFT.

Will test tomorrow to see if mine will work OK from the main board

Drue74, 2022-08-10:
Anyone nail this down yet? Just got printer an it's doing that.

Rick18101965, 2022-08-11:
Nope but going to look this weekend if i have time.  I will look at bq
advising i update firmware.  Got the laptop and wire now and not looking
forward to it.  Pity not ok out of the box.

Rick

Taz21017, 2022-10-01:
I have returned two machines that did exactly what was described. Did updating the firmware correct the issue?