bigtreetech/BIQU-BX #23: Long filenames refuse to start printing

qkdefus, 2022-03-16:
I have an issue where i use a Cura plugin that let me auto re-name sliced files using prefix

[year]_[abbr_machine]_[print_time_hours]h[print_time_minutes]m_[material]_[layer_height]mm_[speed_print]mms_[object_count]x_-_[base_name]_-_[scale]%scale

this results in ofcourse long filenames where my ender 3 pro and ender 2 pro does not have any issue with, both running close to if not latest marlin and octoprint.
however my new biqu bx printer refuse to start printing files with long filenames.

qkdefus, 2022-03-16:
renaming the files with shorter name will print the file just fine.

radek8, 2022-04-17:
In Marlin, enable support for long file names and then compile a new FW.